tjgilbert/ece411 #1: Make Coil Winding Jig

tjgilbert, 2016-10-17:


jmesserpdx, 2016-10-18:
Jeff bought some Lexan for the coil form. Need to cut slot in top piece to feed wire through. Must also have a working drawing for the form at some point but will probably "wing" one on a drill press to get rolling.

tjgilbert, 2016-10-18:
I got some Litz wire in. I'll bring it in tomorrow (10/18)

dahnk, 2016-10-18:
We aren't doing high frequencies now, right? At most around 150kHz? And the amperage should be quite low as well, is this correct? 
If these assumptions are correct I believe we could get away with some simple magnetic wire or a smaller gauge, like 22 or 18 AWG. This is easy enough to get from the LID for our prototyping

jEfEpdx, 2016-10-18:
Yea I am planning to test a small gauge and large gauge magnet wire
tomorrow and characterize the frequency response.

I am not sure on current draw. Less than 2 amps.

closed 2016-10-30 by tjgilbert