tjgilbert/ece411 #6: Characterize Load Profile of various mobiles

jmesserpdx, 2016-10-18:
Need to see how various phones current draws vary with how the battery is charging. Tyler is looking into the iPhone charging profile. Will provide info for designing the  impedance sensing circuitry in the transmitter section.

jmesserpdx, 2016-10-18:
A logging multi-meter would be nice for this. The capstone lab meters have an RS-232 port so I'll look into the manual to see if we can interface to the computer and make pretty graphs.

tjgilbert, 2016-10-18:
I don't think it is necessary to do multiple devices yet. It will be different on our transformer.  I would just categorize this under de-risking.

tjgilbert, 2016-10-19:
**Iphone 4:**

|          | Tx    | Rx    | Wall  |
|----------|-------|-------|-------|
| Charging | 551mA | 736mA |       |
| Standby  | 85mA  | 0     | 0     |
| No Load  | 77mA  | 0     | 0     |
| 90%      | 138mA | 90mA  | 400mA |
| 100%     |       |       | 190mA |

jEfEpdx, 2016-10-19:
So the device draws less current when at 100%. Good! Tyler was this using the kit go test?

dahnk, 2016-10-19:
That's good then we can detect the drop in current, by a voltage drop
across some precision resistor?

jEfEpdx, 2016-10-19:
I have a whole set of 1% 'ers still in the box if they would work

closed 2016-10-30 by tjgilbert