tjgilbert/ece411 #14: Need preliminary design for coil driving circuitry

tjgilbert, 2016-10-25:
Need design for coil driving circuitry, ie.   mosfets, capacitors, resistors,   Part sizes do not matter, we can change that later.

jmesserpdx, 2016-11-10:
This issue continues to persist. I'm going to try some new components tomorrow. This will probably become a blocking issue if we don't sort it by the weekend. 

Have Tested the following configurations

Single transistor pair Class B output stage - Doesn't source enough current and seems to draws to much base current for lab signal generator. 

Parallel transistor Class B Output Stage with Current buffer - Sort of worked. Got around 3Vpp at output coil but waveform was fairly distorted. Only NPN devices seem to work need to determine if design flaw or bad components. 

Sziklai pair Complementary Symmetry - Insufficient output swing @ around 2 Vpp and less than 200mA current draw. Could be promising.

tjgilbert, 2016-11-16:
n-type mosfet array driven by square wave, common source

closed 2016-11-16 by tjgilbert