tjgilbert/ece411 #15: Receiver schematic

dahnk, 2016-11-03:
We have some pics from the meeting on Tuesday in the LID of the output receiver schematic.
pic of this from the whiteboard to be uploaded

Also I have been doing some LTspice simulations and designs.
circuit schematic and simulations to be uploaded

A choice of design must be made and receiver circuit built prototyped and tested

tjgilbert, 2016-11-03:
[ ]  pick a zener diode

jEfEpdx, 2016-11-03:
5v Zeners are pretty common and don't very much across process. True not
true but they are easy to measure. Ended up meeting my dad tonight downtown
to watch the Cubs lift their curse.

tjgilbert, 2016-11-16:
[image]

tjgilbert, 2016-11-16:
[image]

closed 2016-11-16 by tjgilbert