tjgilbert/ece411 #16: choose op amps for filter stage

dahnk, 2016-11-10:
Need to handle frequencies around 125Khz

the LT1022 can handle it, it has an acceptable slew rate

here is the digikey link: 
http://www.digikey.com/product-search/en/integrated-circuits-ics/linear-amplifiers-instrumentation-op-amps-buffer-amps/2556125?k=lt1022

jmesserpdx, 2016-11-10:
If we do a passive filter we may need only a simple transistor CC or CE amp to boost it for the output stage.

dahnk, 2016-11-10:
I agree, currently simulating with active but will also simulate passive
with bjt amplifier following

jmesserpdx, 2016-11-10:
When you simulate BJT see if you can get an offset of 7.2 and 6.6 Volts with the follower. Then the output node can be at Vcc/2

dahnk, 2016-11-12:
I have a bunch of TL740cn JFET op amps and these are working really well, I don't think we need to order anything else, but we can just for more testing, but the slew rate on these is more than sufficient to handle our 125kHz input signal.

closed 2016-11-12 by dahnk