tjgilbert/ece411 #17: Find Appropriate Output transistors

jmesserpdx, 2016-11-10:
The current stable of LM3046s, MPQ2907s, and 3904s may not be relied upon. Too many are burnt out or otherwise damaged. We'd save precious lab time replacing them. 

However : 
Doing a rough calculation if we deliver 6 watts into the 10uH coil at 120 KHz we need to source 900 mA. We could parallel 3904/6's which I will try. We may also consider something like a TIP 41/2 pair but the drive current would demand a current preamp.

To Try:

Parallel 3904/6s 
2N3055 and whatever PNP goes with it
TIP 41/2 or like pair w/ 3904 CC amp

jEfEpdx, 2016-11-10:
Just woke up. Slept 7 hours which is quite a bit more than usual.

I can be on campus by 8, Can't drop the dog off until 7:30. I am looking
through results from last night and will start coming up with a few ideas
to try

tjgilbert, 2016-11-16:
TN0702 [(datasheet)](http://ww1.microchip.com/downloads/en/DeviceDoc/TN0702%20C080813.pdf) x 5  or [RFD16N05LSM9ACT (datasheet)](https://www.fairchildsemi.com/datasheets/RF/RFD16N05LSM.pdf)

closed 2016-11-16 by tjgilbert