tjgilbert/ece411 #18: Characterize Jeff's Coils

jmesserpdx, 2016-11-10:
Maybe we can solder them onto the Adafruit kit and see if they are feasible that way. They seem small but if they work that can be our "market edge". 

Need:
Inductance 
DC Resistance
     - If we end up direct coupling them we may need to see if they can handle the Quiescent current                 
Sensitivity to orientation and placement 
      - Some metric needs to be defined like "50% coupling at 1/4" displacement from center" 
         maybe Qi standard has a metric we can use.

tjgilbert, 2016-11-16:
Do we still want to try to use these?
Litz coils were 20uH self inductance and 140pF parallel capacitance. Resonant frequency ~2.8MHz, max power transfer @ 246kHz

tjgilbert, 2016-12-02:
too late

closed 2016-12-02 by tjgilbert